svartalf/rust-battop #2: Make colored labels more contrast

non-descriptive, 2019-03-11:
https://monosnap.com/file/QgWq6JhUNyRENLxBtXkgHTU3CLkCtp
tested some different themes in terminal and figured out that some of labels is hard to read
hash and temperature are hard to read on basically on any light theme. 
percentage bar value on `solarized dark` and `homebrew` are barely visible.

rpietro, 2019-11-16:
I love this app, but the contrast is really an issue. The white font on light green background is hardly visible, even in your screenshot on the README page. Any chance you could change that. Great work!

lyokum, 2020-05-22:
I moved the battery percentage to the title of the section so that it's more readable, no matter the terminal theme. I also tweaked a few other elements of the UI to make them clearer. You can view the changes at this PR: #10 

[image]