svartalf/rust-battop #3: Debian packaging for Pop!_OS

mmstick, 2019-03-11:
Whenever you're ready, I'll package and add this to our Pop repository. This seems like a very useful tool for us.

svartalf, 2019-05-21:
Hey, @mmstick!

I was hoping to release `battop` for Windows first, but since it is goes to a postponed status too often due to the Windows-specific stuff, I guess it is time to get back to your suggestion about Pop!_OS.
In addition, there was no bug reports for a last few months, so `battop` can be assumed to be stable enough (or probably it's because no one is using it, which is very likely :) ).

I've published the [v0.2.2](https://github.com/svartalf/rust-battop/releases/tag/v0.2.2) release with all fixes available, so nothing should block you now.
Also I still don't know how the build process for Pop!_OS works like, but the release page for each release contains source code (in `.tar.gz` and `.zip`), `.asc` signature and `.sha512` hash for each source archive, hope it will help you.

Let me know if you need any help.

mmstick, 2019-05-21:
Essentially, we create a debian directory to include along with the sources, which contains extra package metadata and a debian-specific Makefile (`debian/rules`). We add rules to create a vendored tarball during the `clean` rule, and use `sbuild` to build the project in a `schroot`. `sbuild` executes the clean rule from the host, then chroots and executes all the rules (incl. the clean rule again). We have a check in the clean rule `if is-chroot` that ensures that the chroot doesn't attempt to remove and re-vendor, as it doesn't have Internet access.

I'll have the debian packaging ready soon.

svartalf, 2019-05-21:
Okay, I understood some of these words :)

What is a source of the "sources" in that case? Are you checking out the "master" branch or using GitHub releases page?
My main concern here is that `master`' HEAD might not be stable, so either you can checkout to the pre-defined tag or download an archive from the [releases page](https://github.com/svartalf/rust-battop/releases) (I intend to create one for each new tag).

mmstick, 2019-05-21:
Sources referring to your source code. We just a debian directory to your repo. We can use tagged commits.

svartalf, 2019-05-21:
@mmstick yeah, tagged commits would be the preferred way, otherwise it might break the build with unexpectedly broken master.