svartalf/rust-battop #5: Windows support

svartalf, 2019-03-11:
This is a tracking issue for Windows support.

Main problem is that [termion](https://crates.io/crates/termion) backend for [tui](https://crates.io/crates/tui) dependency does not support Windows terminals.

Another option for tui - [crossterm](https://crates.io/crates/crossterm) supports Windows and should be considered as a replacement.

svartalf, 2019-03-14:
Possible switch to `crossterm` is blocked by this issue: https://github.com/TimonPost/crossterm/issues/81 (I think), since it is important for `battop` to handle `←`, `→` and `^C` key combinations.

svartalf, 2019-04-11:
Note to myself: blocking PR was closed, nothing should block windows support now.

svartalf, 2019-04-13:
And it is blocked by this PR now: https://github.com/fdehau/tui-rs/pull/150

svartalf, 2019-05-17:
Almost done, blocked only because there is no `tui` crate release with the changes from this PR: https://github.com/fdehau/tui-rs/pull/153

bew, 2021-02-09:
Still blocked? 🙄

svartalf, 2021-02-09:
No, it is not blocked anymore, switch to `crossterm` backend for `tui` should be done, but I don't really have time to do that now.

PRs are more that welcome!

ratijas, 2021-03-12:
Interesting to the point at which I'm almost willing to give it a try.

What was the whole point, again? Terminal UI library, which is being used now, provides too much capabilities that cannot be replicated on Windows, so devs just gave on supporting it, and thus it fails to even compile? So we need to move on to a less capable library which would support the lowest common denominator of all platforms. Am I getting this right?

svartalf, 2021-03-12:
@ratijas the idea is to pick up terminal UI library, which will provide the same interface (or, at least the almost same) as in the current implementation, while also providing support for at least Linux, macOS and Windows. If it will also support *BSD family, that will be a great bonus.

I'm not aware about current state of `termion` crate, but the idea to switch to `crossterm` still stands.

Does this answers your question?