svartalf/rust-battop #6: Problems on Linux

flofriday, 2019-03-12:
I have tested [battop](https://github.com/svartalf/rust-battop) on Linux and I found two issues:
* Temperature doesn't seem to work at all
* Cycles count shows 0 even tough I have used this laptop over 2 years almost every day. ( Maybe it is because I have installed the OS just yesterday? )

Here is a screenshot on Linux Mint 19.1
[image]

svartalf, 2019-03-12:
That's true, temperature might not be available, because driver is not providing it. Just to be sure, can you show the ouput from the following commands?

```
cat /sys/class/power_supply/*/temp
cat /sys/class/power_supply/*/cycle_count
```

flofriday, 2019-03-12:
``` bash
flo@V15:~$ cat /sys/class/power_supply/*/temp
cat: '/sys/class/power_supply/*/temp': No such file or directory
flo@V15:~$ cat /sys/class/power_supply/*/cycle_count
0
```

It can't find the file `cat /sys/class/power_supply/*/temp`

svartalf, 2019-03-13:
Thank you for your help, @flofriday :)

Indeed, your device driver is not providing temperature value, therefore `/sys/class/power_supply/*/temp` file does not exist.

Yet it is odd that it returns `cycle_count = 0` all the time, I was expecting that this field would be missing too in case of unavailable value.
I have created a bug for that case: https://github.com/svartalf/rust-battery/issues/23

svartalf, 2019-03-13:
Temperature graph is now shows "UNAVAILABLE" label in case of temperature data is unavailable :)

Since `cycle_count = 0` bug was created at https://github.com/svartalf/rust-battery/issues/23, I'm going to close that issue.
Thanks once again!

closed 2019-03-13 by svartalf

groutr, 2019-04-11:
I'm having a similar issue with Temperature.  The odd curiosity is that battery temperature is available on my machine via `acpi -t` (Thinkpad T460).  I think this is reporting `/sys/class/thermal/thermal_zone0/temp`