svartalf/rust-battop #7: battop exits when plugging AC back in

razielgn, 2019-05-29:
Running `battop 0.2.2`, installed via `yay -S battop` on ArchLinux `5.1.5-arch1-2-ARCH`.
Battery detected as `SMP DELL GPM0365`.

Steps to reproduce:
1. launch battop
2. plug AC out.
3. wait more than ~2/3 seconds.
4. plug AC back in.
5. battop exits apparently gracefully (running with `RUST_BACKTRACE=1` does not report anything).

stderr collected via `-vvvvv`:
```
2019-05-29T11:50:25+00:00 - ERROR - Error occurred: Battery(Error { source: Os { code: 19, kind: Other, message: "No such device" }, description: None })
Error: Battery(Error { source: Os { code: 19, kind: Other, message: "No such device" }, description: None })
```

svartalf, 2019-05-30:
Hey, @razielgn! I think I've managed to fix this issue, is there any chance you can test it out?
I was not able to reproduce it locally and emulating the `ENODEV` is kinda hard to do properly :)

Since you are using a fresh enough Arch, I think there would be issues with running this binary: [battop-0.2.3-alpha.1.gz](https://github.com/svartalf/rust-battop/files/3236765/battop-0.2.3-alpha.1.gz)

razielgn, 2019-05-30:
I tried it and couldn't reproduce the reported issue! Kudos! :smiley:

svartalf, 2019-05-30:
Yay! Many thanks for such a quick feedback!

I'll publish the new version properly in a next hour or so (into AUR also)

closed 2019-05-30 by svartalf

razielgn, 2019-05-30:
Great! Thank you!

svartalf, 2019-05-30:
And done: https://aur.archlinux.org/packages/battop/

commit eee519cb (2019-05-30): Release v0.2.3 (closes #7)